aus-plant-phenomics-network/phenomate #8: Dockerise Phenomate

harryle95, 2025-07-31:
- [ ] Docker compose scripts with - celery service, front end service, and back end service.
- [ ] Allow users to specify environment variables: 
    - [ ] Admin Username, Email, Password
    - [ ] Ports: frontend, backend, postgresql, celery
    - [ ] Default directory for file viewing
    - [ ] Default project root directory
    - [ ] Postgresql configs
    - [ ] Celery configs 
- [ ] Expose host filesystem - need to test with windows etc

harryle95, 2025-08-06:
Need to complete #10 task for setting up sending notif using admin email.

jcbowden-appn, 2025-08-14:
N.B. Perhaps no option to be given for deletion of bin files

jcbowden-appn, 2025-08-28:
check with George for docker-compose override file for drive binding windws/linux

gsainsbury86, 2025-08-28:
[Some info about override compose files here.](https://docs.docker.com/compose/how-tos/multiple-compose-files/merge/)

The short version is that you can specify _just_ the changes you might want for a specific environment. E.g. dev or Windows vs Linux etc.

jcbowden-appn, 2025-08-29:
Hi George, I've been testing the override system in docker compose and it works well. It gives me options to setup the docker images that differ for WSL2 and Powershell on Windows, and Linux. If you need something different for Mac then we should work something out about how that differs from Linux. I currently cannot test on Mac.

jcbowden-appn, 2025-09-03:
My current testing of use-case-2 STAAS drive access has resulted in the thinking that the virtual file driver used to access a filesystem used by the Phenomate code fails to find a path if it is a Windows network drive. I have tried various options to get the network drive accessible. My initial problem being that WSL2 is not automatically mounting a mapped drive into the WSL filesystem (which is a documented method, and allows the ```C:``` drive to be accessible within the containers at ```\mnt\c```).

jcbowden-appn, 2025-11-20:
Running Phenomate using Docker on Windows has been deemed a bad idea, mostly due to the unusual file browser component (VFS) that is being used.
The system running on Linux is working okay. Issue found with database storage for GUI field data pre-population could be done better.