aus-plant-phenomics-network/phenomate #9: Collect more information for project construction

harryle95, 2025-07-31:
The current setup uses `year_summary_internal_researcher_organisation`. Allows for more fields to be collected from the UI. 

For instance - year, month, day, etc...

jcbowden-appn, 2025-08-14:
add: project number, plant species, and note section

jcbowden-appn, 2025-09-17:
To add extra information to the frontend and pass it through to the backend requires a complex web of changes to be made to various files and parts of the application source code. The only good thing about the modification process is that when changing the schema in the backend, the frontend will be updated using the typescript update using:

```bash
npm run sync-backend
#  sync-backend is defined in project.json as:
# sync-backend": "openapi-typescript http://127.0.0.1:8000/api/openapi.json -o ./src/lib/api/v1.d.ts"
```

The Django backend has a schema system to present the frontend with the correct data structures that it expects.

As an example being for creating a 'project' using the GUI - from file ```backend\projects\dto.py```

```python
class ProjectCreateSchema(Schema):
    year: int
    summary: str
    template: str | None = None
    internal: bool = True
    researcherName: str | None = None
    organisationName: str | None = None
    root: str | None = None
```

The above schema is exposed in Django ```backend\project\api.py```
```python
from appm import ProjectManager  # This is the project template module
from backend.project.dto import (
    ProjectCreateSchema,
...
)
...
@router.post(
    "/",
    response=ProjectListSchema,
    summary="Create a new project",
)
def create_project(request: HttpRequest, data: ProjectCreateSchema) -> Project:
    researcher, rstatus = (
        Researcher.objects.get_or_create(name=data.researcherName)
        if data.researcherName
        else (None, None)
    )
    organisation, ostatus = (
        Organisation.objects.get_or_create(name=data.organisationName)
        if data.organisationName
        else (None, None)
    )
    root = data.root if data.root else getattr(settings, "DEFAULT_ROOT_FOLDER", "/Project")
    manager = ProjectManager.from_template(
        root=root,
        year=data.year,
        summary=data.summary,
        internal=data.internal,
        researcherName=researcher.name if researcher else None,
        organisationName=organisation.name if organisation else None,
        template=data.template,
    )
    project = Project.objects.create(
        year=data.year,
        summary=data.summary,
        internal=data.internal,
        researcher=researcher,
        organisation=organisation,
        location=str(manager.location.absolute()),
    )
    manager.init_project()
    return project
```

Within the appm module:
```python
@classmethod
    def from_template(
        cls,
        root: str | Path,
        year: int,
        summary: str,
        internal: bool = True,
        template: str | Path | dict[str, Any] | None = None,
        researcherName: str | None = None,
        organisationName: str | None = None,
    ) -> ProjectManager:
        """Create a ProjectManager based on template and meta information

        Args:
            root (str | Path): parent directory - where project is stored
            template (str | Path | dict[str, Any]): path to template file or the template content.
            year (int): meta information - year
            summary (str): meta information - summary
            internal (bool, optional): meta information - internal. Defaults to True.
            researcherName (str | None, optional): meta information - researcher name. Defaults to None.
            organisationName (str | None, optional): meta information - organisation name. Defaults to None.

        Returns:
            ProjectManager: ProjectManager object
        """
        if isinstance(template, str | Path):
            metadata_path = Path(template)
            metadata_path = validate_path(template)
            with metadata_path.open("r") as file:
                metadata = yaml.load(file)
        elif isinstance(template, dict):
            metadata = template
        elif not template:
            metadata = deepcopy(DEFAULT_TEMPLATE)
        else:
            raise TypeError(
                f"Unexpected type for template: {type(template)}. Accepts str, dict or None"
            )
        metadata["meta"] = {
            "year": year,
            "summary": summary,
            "internal": internal,
            "researcherName": researcherName,
            "organisationName": organisationName,
        }
        return cls(root=root, metadata=metadata)
```

jcbowden-appn, 2025-09-19:
Casual comment: Updating this code is not a straightforward process. I will try to add a single new field "siteName" and if that works then we can work from there.

One improvement would be to split the appn-project-manager code into parts:
- Deals with filename decomposition
- Deals with directory path for output.

jcbowden-appn, 2026-01-15:
Platform Name (e.g. phenomate_1), and Site Name (e.g. Horsham) fields have been added and integrated with `appm` package backend code and templates used for data placement.

New frontend:
[image]

closed 2026-01-15 by jcbowden-appn