aus-plant-phenomics-network/phenomate #19: change basename regex in tempalte.yaml

sicoughlin, 2025-08-13:
Basename that template.yaml accepts could be improved for our amiga projects.

We'd prefer it if the basename it accepted looked like this:

20250623-022409_[basename]_[instrumentname].bin 
I dont think we'll need a file counter in the name, since the timestamp will give us that information. 
Also I don't think the trial number will be useful in the basename, since that will be obvious from the different date-time in the basename.

jcbowden-appn, 2025-08-14:
I'm getting the latest data in the latest directory structure today. So, i'll test what is happening and work from there.

jcbowden-appn, 2025-08-19:
The data received has the following directory and file name structure:

The structure of the filename is not very consistent. 
- The date (which is repeated as `2025-08-14` and `0814`), the `test2` string and the final `0` run number (for lack of a better name) are equivalent.
- The string after the first date string, between the first two `_` characters are slightly different between the instruments. e.g. `06-23-10_137493` could encode a time of `06:23:10.137493` in `hr:min:sec.ms` format.
- The `_0814_` seems to be a repeat of the date
- The `test2` string is consistent, however is repeated more sometimes in some instrument output and not others.
- 
```bash
14_aug/canbus/2025-08-14_06-23-10_137493_0814_test2_canbus_0_MotorState.csv

14_aug/imu1/2025-08-14_06-25-53_306635_0814_test2_imu1_0.bin
14_aug/imu1/2025-08-14_06-25-53_306635_0814_test2_imu1_0.csv
14_aug/imu1/2025-08-14_06-25-53_306635_0814_test2_imu1_0_GNSS.csv

# Here there is a choice of files:
14_aug/jai1/2025-08-14_06-24-53_448807_0814_test2_jai1_0.bin
14_aug/jai1/2025-08-14_06-23-45_953704_0814_test2_jai1_0.bin

14_aug/Lidar2/2025-08-14_06-25-48_119291_0814_test2_Lidar2_0.bin
```
The format may be clearer in the Resonate repositories. I will take a look.

@sicoughlin and/or @gsainsbury86 , can possibly shed some light on how the files are being named.

jcbowden-appn, 2025-08-19:
The previous data recieved had a format:
```
20250623-022409_waite_jai_trial-1_camera-1.bin
```
which was matched in the appn-project-manager\examples\template.json file as
```json
file:
  "*":
    sep: "_"
    default:
      procLevel: raw
    components:
      - sep: "-"
        components:
          - ['date', '\d{8}']
          - ['time', '\d{6}']
      - ['site', '[^_.]+']
      - ['sensor', '[^_.]+']
      - ['trial', '[^_.]+']
      - name: 'procLevel'
        pattern: 'T0-raw|T1-proc|T2-trait|raw|proc|trait'
        required: false
```

jcbowden-appn, 2025-08-20:
Timestamps and filenames are created in the ```appn-instrument-interfaces`` repository under the ```instruments/sensor name/<sensorname>_inst.py``` folder/file:

```bash
APPN/repos/RESONATESYSTEMS$ grep -RnI "%Y-%m" .
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/Instruments/IMU/INS_401_ethernet.py:231:                datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S')))
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/Instruments/IMU/INS_401_ethernet.py:243:                datetime.datetime.now().strftime('%Y-%m-%d %H:%M:%S')))
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/Instruments/IMU/imu_inst.py:68:        timestamp = datetime.now().strftime("%Y-%m-%d_%H-%M-%S_%f")
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/Instruments/LiDAR/lidar_inst.py:149:        timestamp = datetime.now().strftime("%Y-%m-%d_%H-%M-%S_%f")
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/Instruments/HyperSpecSensor/hyperspec_inst.py:85:        timestamp = datetime.now().strftime("%Y-%m-%d_%H-%M-%S_%f")
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/Instruments/OAK-D/oak_d_inst.py:108:        timestamp = datetime.now().strftime("%Y-%m-%d_%H-%M-%S_%f")
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/Instruments/CANBUS/canbus_inst.py:94:        timestamp = datetime.now().strftime("%Y-%m-%d_%H-%M-%S_%f")
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/Instruments/JAI/jai_inst.py:85:        timestamp = datetime.now().strftime("%Y-%m-%d_%H-%M-%S_%f")
./resonatesystems-rs24005-appn-instrument-interfaces-dca2e05b997d/CHANGELOG.md:15:- The timestamp format in all filenames has been changed to %Y-%m-%d_%H-%M-%S_%f
```

jcbowden-appn, 2025-08-20:
There has been a proposal (via email) to set the file format string in a ```.env``` file within the ```appn-instrument-interface``` repository and to save some metadata at the start of collection that:

> Details such as the following would be useful:
The software version being used (git tag and/or commit hash).
The string format being used for the filename.
The runtime system being used (Python version, Brain OS version?).

> Things like the string format can be read from an .env file and used everywhere in the codebase.

> The saved metadata (in JSON format, as a universally accessible starting point) can then be used in upstream processing to read the files and extract the metadata saved in the filename, and in data provenance logs.

jcbowden-appn, 2025-08-25:
### Some notes on the template
The template for extracting metadata from the input protobuf binary file filenames is selected from JSON based files found in the appm repository: ```appn-project-manager\examples\```  

There is also a default template read from file ```appn-project-manager\appm\default.py```
  
The actual template used for a conversion is saved in the output directory selected by the user when transfer occurs. This file is named ```metadata.yaml``` and has a ```version``` field that is read from ```appn-project-manager\appm\__version__.py```

So, this means that a project version should currently be updated in:  
- ```pyproject.toml```
- ```\appm\__version__.py```

jcbowden-appn, 2025-08-26:
There is a function in ```appm\manaer.py``` line 90 ```save_metadata.py``` that adds/rewrites the data['version'] element of the data saved as ```metadata.yaml``` to the value found in file ```appm\__version__.py```.
   
This could be changed to the value found in the ```template.yaml``` file used to populate the metadata which is (or can be) chosen by the user (or otherwise sourced from a file ```appm\default.py```.

### Todo 
- Think about a better way to denote provenance of the version value.

jcbowden-appn, 2025-08-27:
bump-my-version is now being used in the ```appm``` and ```phenomate-core``` repos.

jcbowden-appn, 2025-11-20:
As an extension to this, there is a task that is to add a 'preview' of the directory structure as GUI fields are filled in by a user when creating a project.